lewisxhe/SensorLib #2: Question: QMI8658 example gets 1/2 of true acceleration value when gyro is disabled

lyusupov, 2023-05-03:
**Sketch**: [QMI8658_GetDataExample](https://github.com/lewisxhe/SensorsLib/blob/master/examples/QMI8658_GetDataExample/QMI8658_GetDataExample.ino)

**Normal use case**:
when both gyro and accel are enabled - the getAccelerometer() function gives correct acceleration value

**Issue**:
when gyro is disabled like that:

```
#if 0
    qmi.configGyroscope(
        /*
        * GYR_RANGE_16DPS
     
< --- skipped --->

    // In 6DOF mode (accelerometer and gyroscope are both enabled),
    // the output data rate is derived from the nature frequency of gyroscope
    qmi.enableGyroscope();
#endif
```

- then reported acceleration value is a half of true value.

The workaround is to enable the gyro all the time like the [SoftRF did](https://github.com/lyusupov/SoftRF/commit/8dee1804e3e952532f03efb930c3bc1f45720abe) ... 
But could we avoid that and get correct acceleration value regardless if the gyro is on or off ?

lewisxhe, 2023-05-04:
Give me some time, I'm currently on vacation

LazaroFilm, 2023-07-12:
This may be related to the fact that the Accelerometer's ODR has different values when used alone than with the gyro on. see from documentation.

[image]

lewisxhe, 2024-09-23:
@lyusupov 

Linar .

I tested QMI today and found no change in acceleration value when turning on and off the gyroscope.

Tested turning on acceleration alone
[image]

Turning on acceleration and gyroscope
[image]

From the manual, we know that the output frequency of turning on acceleration alone comes from the internal oscillator.
When turning on acceleration and gyroscope at the same time, the output frequency of both is changed to the output frequency of the gyroscope. My personal understanding is that only the output frequency is changed, and the data value will not be changed.

lewisxhe, 2024-09-23:
No, I tested it differently than you did. I configured the gyroscope and then disabled it.
After testing, this may be related to the gyroscope's self-test.

lewisxhe, 2024-09-23:
After removing the self-test, enabling acceleration alone now works fine, the self-test usage was wrong before.

lyusupov, 2024-09-23:
Thank you, Lewis!

I am unable to verify the sketch right now...
If you are sure that the issue is no longer a factor after your fix - please, feel yourself free to close this ticket.

lewisxhe, 2024-09-23:
OK. I'll close it first. Feel free to open it if you have any questions.

closed 2024-09-23 by lewisxhe