lewisxhe/SensorLib #3: QMI8658 AttitudeEngine

LazaroFilm, 2023-08-21:
Has anyone succeeded in using the AttitudeEngine internal fusion from the QMI8658?

lyusupov, 2023-08-22:
AttitudeEngine in QMI8658 has been inherited with a license acquired from Fairchild...
Are there some useful applications available for FIS1100 AttitudeEngine ? I don't know any.

LazaroFilm, 2023-08-22:
I'm confused. I thought AttitudeEngine was outputting a fused sensor quaternion from the QMI8658. Does it not do that?

lyusupov, 2023-08-22:
**AttitudeEngine** is a registered **trademark** of Fairchild Semiconductor International _(acquired by ON Semiconductor Corporation in 2016)_.

https://www.onsemi.com/download/application-notes/pdf/an-5083.pdf

jake653, 2024-02-29:
The QMI8658C data sheet references Fusion and the XKF3 an external library for ARM processors (XKF3, developed in 2005 has been described to do auto calibration and plotting?)
https://www.qstcorp.com/upload/pdf/202202/%EF%BC%88%E5%B7%B2%E4%BC%A0%EF%BC%89QMI8658C%20datasheet%20rev%200.9.pdf

 the QMI8658A data sheet included in this archive doesn't reference fusion.  I see some header files (waveshare "RP2040-Touch-LCD-1.28" and ESP32-S3 version have the best example source code I have found) that references fusion:
extern void QMI8658_read_ae(float quat[4], float velocity[3]);
unsigned char const *attitudeEngineData;    /*! \brief Raw sample buffer. */

C source includes
void QMI8658_read_ae(float quat[4], float velocity[3])

The Fairchild/On Semi XKF3 seem to have special licensing requirements, I'm hazy on what gets done internally and what'd doe by XKF3.

LazaroFilm, 2024-02-29:
Yeah this has been frustrating. I even reached out to the manufacturer and have not received any answers.

lewisxhe, 2024-04-18:
Attitude engine, I have contacted chip manufacturers for this many times. They can provide it but cannot make it open source, which may involve patent issues. Question will be closed

closed 2024-04-18 by lewisxhe