lewisxhe/SensorLib #5: QMC6310_GetDataExample compile on Arduino Nano

Rosscoetrain, 2023-11-11:
Following error occurs when trying to compile on an arduino nano

Where is <vector>?

and how can I compile it with only the QMC6310 driver.

Arduino: 1.8.19 (Linux), Board: "Arduino Nano, ATmega328P (Old Bootloader)"

In file included from /home/ross/Arduino/libraries/SensorLib/src/bosch/BoschParse.cpp:30:0:
/home/ross/Arduino/libraries/SensorLib/src/bosch/BoschParse.h:36:10: fatal error: vector: No such file or directory
 #include <vector>
          ^~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano.

lyusupov, 2023-11-12:
https://github.com/lewisxhe/SensorLib/blob/master/library.json#L30

lewisxhe, 2024-04-18:
As mentioned above

closed 2024-04-18 by lewisxhe