lewisxhe/SensorLib #13: Native support for ESPHome?

dala318, 2024-09-12:
First of all, great library!
I'm a bit noob to libraries, includes etc. but would it be possible to make this library fully compatible with ESPHome out of the box?
As far as I have understood ESPHome supports Arduino i2c libraries "out of the box" to be used in custom integrations [Custom I²C Device](https://esphome.io/custom/i2c.html), but I'm struggling to get it work.

One thing that hits me is the include pathes, are they relative or absolute etc. but that I guess is only trial and error and a bit of patience.

The other thing that troubles me is that the library uses printf in some places, while ESPHome would like it through their own ESP_LOGx.

Then I also struggle to understand how an instance of a selected library will get the handle to the i2c instance. It seem in the library that it imitates it by its own, while ESPHome idea is to initiate that externally and only pass the handle to it for each integration.

I also created a [feature request](https://github.com/esphome/feature-requests/issues/2880) in ESPHome for this, don't know what the easiest way to do it is...

dala318, 2024-09-12:
Or is it this begin method I should use and provide a function pointer to my native ESPHome i2c component?
[SensorCommon.tpp#L234](https://github.com/lewisxhe/SensorLib/blob/master/src/SensorCommon.tpp#L234)

lewisxhe, 2024-09-13:
It should be compatible. I remember that you can directly pass in the Wire handle. You can try it. I don't use esp home.

dala318, 2024-09-13:
Thanks for confirming that part, you don't happen to have an example how one would define send/receive functions and pass as arguments to that begin method?

lewisxhe, 2024-09-13:
You can see here that XPowersLib and SensorLib are written using the same method, so they are basically the same https://github.com/esphome/issues/issues/4168

dala318, 2024-09-13:
What all examples do (and I suppose that's because it the correct way) is that they let the integration initiate the i2c bus by the [begin(PLATFORM_WIRE_TYPE &w, uint8_t addr, int sda, int scl)](https://github.com/lewisxhe/SensorLib/blob/master/src/SensorCommon.tpp#L71) method and I suppose it works. The only downside of that is that you have to put the SDA and SCL pins in your code (or pass as arguments to the setup of your "Component").

What I was hoping to achieve was to have my component extend both PollingComponent and I2CDevice which then let me configure the i2c bus in a separate section in ESPHome yaml.
I've come a bit on the way now, the trick was to define the read/write callback as static (and not as class functions)

Thanks once again for the support!

lewisxhe, 2024-09-13:
For ESP, you only need to pass in the Wire handle, SDA and SCL are not necessary. You only need to initialize the Wire before begin, and then you don't need to pass the pin to them.

closed 2024-09-13 by dala318

lewisxhe, 2024-09-18:
> Still need som cleanup and tweaking but have got it working by the above hints.
> Kept with the Arduino framework for now and double SDA/SCL definitions.
> dala318/esphome-qmi8658<https://github.com/dala318/esphome-qmi8658> for anyone else interested. (no ready "use as is" external_component at the moment but can copy stuff from it if you want)