lewisxhe/SensorLib #14: QMI8658_GetDataExample accelaration values are not zero when in rest

Jeroen88, 2024-09-21:
Using a Lolin S3 mini pro with an integrated QMI8658 sensor and the Arduino IDE, the sensor values are not zero when the sensor is not moving. When I move the sensor (or actually the board with the sensor on it) the values change. So the sensor seems to work.

The only modification I made to the test program is setting SDA to 12 and SCL to 11.

Should the sensor be calibrated? I am also not sure if this problem is caused by an error in the library or just a bad sensor.

The output of the test program is:

```
21:13:27.479 -> [initImpl] FW Version :0x2F0201
21:13:27.479 -> 
21:13:27.479 -> [initImpl] USID :2150296DDACB
21:13:27.479 -> 
21:13:27.479 -> Device ID:7B
21:13:27.511 -> CTRL1: REG:0x02 HEX:0x60  BIN:0b1100000
21:13:27.511 -> CTRL2: REG:0x03 HEX:0x93  BIN:0b10010011
21:13:27.511 -> CTRL3: REG:0x04 HEX:0xA3  BIN:0b10100011
21:13:27.511 -> CTRL4: REG:0x05 HEX:0x00  BIN:0b0
21:13:27.511 -> CTRL5: REG:0x06 HEX:0x71  BIN:0b1110001
21:13:27.511 -> CTRL6: REG:0x07 HEX:0x00  BIN:0b0
21:13:27.511 -> CTRL7: REG:0x08 HEX:0x03  BIN:0b11
21:13:27.511 -> CTRL8: REG:0x09 HEX:0x80  BIN:0b10000000
21:13:27.511 -> CTRL9: REG:0x0A HEX:0x00  BIN:0b0
21:13:27.511 -> 
21:13:27.511 -> FIFOCTRL: REG:0x14 HEX:0x00  BIN:0b0
21:13:27.511 -> Read data now...
21:13:27.900 -> {ACCEL: 0.63,0.23,0.72}
21:13:27.900 -> {GYRO: -1.90,-0.12,-0.47}
21:13:27.900 -> 				 > 53  28.78*C
21:13:27.996 -> {ACCEL: 0.63,0.23,0.72}
21:13:27.996 -> {GYRO: -1.60,-0.54,-0.14}
21:13:28.028 -> 				 > 146  28.55*C
21:13:28.124 -> {ACCEL: 0.63,0.23,0.73}
21:13:28.124 -> {GYRO: -1.70,-0.62,-0.12}
21:13:28.124 -> 				 > 238  28.57*C
21:13:28.220 -> {ACCEL: 0.63,0.23,0.72}
21:13:28.220 -> {GYRO: -1.48,0.54,-1.18}
21:13:28.220 -> 				 > 330  28.57*C
21:13:28.317 -> {ACCEL: 0.63,0.23,0.72}
21:13:28.317 -> {GYRO: -1.88,-0.99,-0.60}
21:13:28.317 -> 				 > 422  28.56*C
21:13:28.412 -> {ACCEL: 0.63,0.23,0.72}
21:13:28.413 -> {GYRO: -1.60,0.80,-0.96}
21:13:28.413 -> 				 > 514  28.55*C
21:13:28.541 -> {ACCEL: 0.63,0.23,0.72}
21:13:28.541 -> {GYRO: -2.06,-1.09,-0.92}
21:13:28.541 -> 				 > 607  28.55*C
21:13:28.637 -> {ACCEL: 0.63,0.23,0.72}
21:13:28.637 -> {GYRO: -1.53,0.72,-0.67}
21:13:28.637 -> 				 > 699  28.59*C
```

I have two of those boards and both give non (near) zero, but very different values:
```
21:19:05.549 -> [initImpl] FW Version :0x2F0201
21:19:05.549 -> 
21:19:05.549 -> [initImpl] USID :2F4C296DDACB
21:19:05.549 -> 
21:19:05.549 -> Device ID:7B
21:19:05.549 -> CTRL1: REG:0x02 HEX:0x60  BIN:0b1100000
21:19:05.549 -> CTRL2: REG:0x03 HEX:0x93  BIN:0b10010011
21:19:05.549 -> CTRL3: REG:0x04 HEX:0xA3  BIN:0b10100011
21:19:05.549 -> CTRL4: REG:0x05 HEX:0x00  BIN:0b0
21:19:05.549 -> CTRL5: REG:0x06 HEX:0x71  BIN:0b1110001
21:19:05.549 -> CTRL6: REG:0x07 HEX:0x00  BIN:0b0
21:19:05.549 -> CTRL7: REG:0x08 HEX:0x03  BIN:0b11
21:19:05.549 -> CTRL8: REG:0x09 HEX:0x80  BIN:0b10000000
21:19:05.549 -> CTRL9: REG:0x0A HEX:0x00  BIN:0b0
21:19:05.549 -> 
21:19:05.549 -> FIFOCTRL: REG:0x14 HEX:0x00  BIN:0b0
21:19:05.549 -> Read data now...
21:19:06.056 -> {ACCEL: 0.53,0.04,0.81}
21:19:06.056 -> {GYRO: 2.09,2.31,-0.38}
21:19:06.056 -> 				 > 87  26.25*C
21:19:06.152 -> {ACCEL: 0.53,0.04,0.80}
21:19:06.152 -> {GYRO: 2.57,1.76,0.06}
21:19:06.152 -> 				 > 181  26.09*C
21:19:06.248 -> {ACCEL: 0.53,0.04,0.81}
21:19:06.248 -> {GYRO: 2.31,1.66,-1.42}
21:19:06.248 -> 				 > 275  26.11*C
21:19:06.345 -> {ACCEL: 0.53,0.04,0.81}
21:19:06.376 -> {GYRO: 2.51,1.75,-0.50}
21:19:06.377 -> 				 > 369  26.12*C

```
So it might also be the (bad) quality of the sensor.

lyusupov, 2024-09-21:
return values of `getAccelerometer()` method are rated relative to **G** (9.81 m/c^2) value. 

https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkIII.-Screen-pages#g-load

[image]

Jeroen88, 2024-09-22:
@lyusupov Thank you Linar, you've taught me something new. Closing the issue!

closed 2024-09-22 by Jeroen88

LazaroFilm, 2024-09-23:
The gyroscope not showing zeros when at rest is more annoying. I have seems some gyro calibrations that would take the gyro values over time, and subtract them as offset values.

LazaroFilm, 2024-09-23:
There is a really well thought out method in the Fusion library by xioTechnologies here: 

https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionOffset.c 
Look at `FusionOffsetUpdate()` and `gyroscopeOffset`
`FusionVectorAdd` multiplies values from tow sets of XYZ values and `FusionVectorMultiplyScalar` multiplies the each values by a given scale 

It creates a self adjusting gyroscope offset. It waits for low gyroscope movement and when the initial values are below the threshold, it considered it  being at rest and will slowly offset the giro values with a scaled down gyro value to slowly bring it to 0. That means that if your gyro drifts in different directions each time, this will learn from it and bring the drift back down. It's nifty in some cases but may not work in all situations (ie if the gyro is sometimes moving at a very slow constant rate and not held by a human, the calibration may overcompensate, but that's a fringe scenario (and you could still set a Lower threshold) you can also look for accelerometer values and make sure they're not moving too much before enabling that calibration.

Jeroen88, 2024-09-23:
@LazaroFilm thank you for the link, I will give it a try!