lewisxhe/SensorLib #16: SensorQMI8658.hpp fails to compile in pure esp-idf projects

Vinzenz82, 2024-12-07:
Hi Coders,

Thanks for that nice lib. I use this lib in a esp-idf VsCode project without any Arduino stuff. 
Unfortunatley the file SensorQMI8658.hpp refers at some line of codes to:

`Serial.println(..)`

without using any ifdef switch like:

```c
#if defined(ARDUINO)
            Serial.printf(...);
#else
            printf(...);
#endif`
```
For example here:
https://github.com/lewisxhe/SensorLib/blob/e3579bc8ec2a2f329c08d1bb3e28ef3ba5b1a205/src/SensorQMI8658.hpp#L1764C13-L1764C19

lyusupov, 2024-12-07:
https://github.com/lewisxhe/SensorLib/actions/runs/7490953952/job/20391015965

lewisxhe, 2024-12-07:
I have not tested the actual effect of QMI in ESP-IDF. The problem you mentioned has been removed.

Vinzenz82, 2024-12-08:
Thanks. It's working.

closed 2024-12-08 by Vinzenz82