lewisxhe/SensorLib #17: BHI360 Sensor Support

federicaventriglia, 2024-12-10:
Any chance this library could also be used for the BHI360 module by Bosch?

lewisxhe, 2024-12-11:
I don't have BHI360, this is based on [BHY2_SensorAPI]( https://github.com/boschsensortec/BHY2_SensorAPI), so it should be possible, but I can't help

federicaventriglia, 2024-12-11:
@lewisxhe I'm trying to do that, but I'm wondering, in your project are you using a GPIO extender? I see the read through spi being references to XL9555

```   
 // Read 16-bit gpio input status
    uint16_t read()
    {
        uint16_t val = 0x0;
        readRegister(XL9555_CTRL_INP0, (uint8_t *)&val, 2);
        return val;
    }
```

lewisxhe, 2024-12-13:
No extensions

federicaventriglia, 2025-01-06:
I've had a go at creating a wrapper around the  BHI360 sensor, which (at least for the initialization) is very similar to the BHI26AP with some minor changes around the firmware selection. One thing I seem to be stuck with is that my sensor is always in "**sleeping**" mode, for some reason. Did anyone run into this issue before? I might be doing something wrong either in my wiring or in my register selection, but I couldn't find anything in the datasheet related to it. 

What I see after my firmware is uploaded is:

```
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:1076] BHI260/BHA260 found. Product ID read 89
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:581] Upload Firmware ...
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:602] Loading firmware into RAM.
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:603] upload size = 127228
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:606] Loading firmware into RAM Done
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:622] Booting from RAM.
[D][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:472] Boot successful. Kernel version 5991.
[I][~/Documents/Arduino/libraries/SensorLib/src/SensorBHI260AP.hpp:1130] Boot successful. Kernel version 5991.
[I][~/Documents/Arduino/libraries/SensorLib/src/bosch/BoschParse.cpp:184] [META EVENT WAKE UP] Firmware initialized. Firmware version 5991
[I][~/Documents/Arduino/libraries/SensorLib/src/bosch/BoschParse.cpp:184] [META EVENT] Firmware initialized. Firmware version 5991
```

then when I try to print the status I don't see any warnings (according to what I found in the application note [here](https://www.bosch-sensortec.com/media/boschsensortec/downloads/application_notes_1/bst-bhi260_bhi360-an002.pdf)) 

```
Initializing the sensor successfully!
Product ID     : 89
Kernel version : 5991
User version   : 5991
ROM version    : 5166
Power state    : sleeping  <------- here 
Host interface : SPI
Feature status : 0x4a
Boot Status : 0x38: 	No flash installed. 
	Host interface ready. 
	Firmware verification done. 
```

and even virtual sensors show up when I use `bhy2_get_virt_sensor_list(bhy2)`:

```
Host interrupt control
-- Wake up FIFO enabled.
-- Non wake up FIFO enabled.
-- Status FIFO enabled.
-- Debugging enabled.
-- Fault enabled.
-- Interrupt is active high.
-- Interrupt is level triggered.
-- Interrupt pin drive is push-pull.
Present sensors: 
1 - Accelerometer passthrough
3 - Accelerometer uncalibrated
4 - Accelerometer corrected
5 - Accelerometer offset
6 - Accelerometer corrected wake up
7 - Accelerometer uncalibrated wake up
10 - Gyroscope passthrough
12 - Gyroscope uncalibrated
13 - Gyroscope corrected
14 - Gyroscope offset
15 - Gyroscope wake up
16 - Gyroscope uncalibrated wake up
28 - Gravity vector
29 - Gravity vector wake up
31 - Linear acceleration
32 - Linear acceleration wake up
37 - Game rotation vector
38 - Game rotation vector wake up
43 - Orientation
44 - Orientation wake up
136 - Low Power Step counter
137 - Low Power Step detector
142 - Low Power Any motion
153 - Multi Tap Detector
154 - Activity recognition for Wearables
156 - Low Power Wrist Gesture wake up
158 - Low Power Wrist Wear wake up
159 - Low Power No Motion wake up
Total 28 Sensor online .
```

but after this, when I try to enable the interrupts for Accel And Gyro for example, nothing ever happens. 

```
    bhy.configure(SENSOR_ID_ACC_PASS, sample_rate, report_latency_ms);
    bhy.configure(SENSOR_ID_GYRO_PASS, sample_rate, report_latency_ms);
    bhy.onResultEvent(SENSOR_ID_ACC_PASS, accel_process_callback);
    bhy.onResultEvent(SENSOR_ID_GYRO_PASS, gyro_process_callback);
```

lewisxhe, 2025-01-07:
I don't have this, can't help

closed 2025-01-07 by lewisxhe

federicaventriglia, 2025-01-07:
I was just wondering if you had run into this type of issue on a BHI260, since they're quite similar..

lewisxhe, 2025-01-07:
No, after initialization is complete, the sensor will be turned on after registering the sensor event. BHI360 I really can't do anything

lewisxhe, 2025-01-07:
Maybe you should ask for support from Bose Sensors, although their support sucks. I had a lot of problems writing this and they didn't give me any support.

federicaventriglia, 2025-01-09:
Yeah I have been having the same issues you mentioned with customer support. Just one question if I may, when you say 

> the sensor will be turned on after registering the sensor event 

do you mean when I issue the following instructions, or somewhere in the initialization of the sensor (assuming it's BHI260)

```
    bhy.configure(SENSOR_ID_ACC_PASS, sample_rate, report_latency_ms);
    bhy.onResultEvent(SENSOR_ID_ACC_PASS, accel_process_callback);

```

lewisxhe, 2025-01-10:
Correct! Below is my test log

```
Initializing the sensor successfully!
Product ID     : 89
Kernel version : 5991
User version   : 5991
ROM version    : 5166
Power state    : sleeping
Host interface : I2C
Feature status : 0x4a
Boot Status : 0x38: 	No flash installed. 
	Host interface ready. 
	Firmware verification done. 
Virtual sensor list.
Sensor ID |                          Sensor Name |  ID | Ver |  Min rate |  Max rate |
----------+--------------------------------------+-----+-----+-----------+-----------|
        1 |            Accelerometer passthrough | 205 |   1 |    1.5625 |  400.0000 |
        3 |           Accelerometer uncalibrated | 203 |   1 |    1.5625 |  400.0000 |
        4 |              Accelerometer corrected | 241 |   1 |    1.5625 |  400.0000 |
        5 |                 Accelerometer offset | 209 |   1 |    1.0000 |    1.0000 |
        6 |      Accelerometer corrected wake up | 192 |   1 |    1.5625 |  400.0000 |
        7 |   Accelerometer uncalibrated wake up | 204 |   1 |    1.5625 |  400.0000 |
       10 |                Gyroscope passthrough | 207 |   1 |    1.5625 |  400.0000 |
       12 |               Gyroscope uncalibrated | 244 |   1 |    1.5625 |  400.0000 |
       13 |                  Gyroscope corrected | 243 |   1 |    1.5625 |  400.0000 |
       14 |                     Gyroscope offset | 208 |   1 |    1.0000 |    1.0000 |
       15 |                    Gyroscope wake up | 194 |   1 |    1.5625 |  400.0000 |
       16 |       Gyroscope uncalibrated wake up | 195 |   1 |    1.5625 |  400.0000 |
       28 |                       Gravity vector | 247 |   1 |    1.5625 |  400.0000 |
       29 |               Gravity vector wake up | 198 |   1 |    1.5625 |  400.0000 |
       31 |                  Linear acceleration | 246 |   1 |    1.5625 |  400.0000 |
       32 |          Linear acceleration wake up | 197 |   1 |    1.5625 |  400.0000 |
       37 |                 Game rotation vector | 252 |   1 |    1.5625 |  400.0000 |
       38 |         Game rotation vector wake up | 200 |   1 |    1.5625 |  400.0000 |
       48 |                        Tilt detector | 236 |   1 |    1.0000 |    1.0000 |
       50 |                        Step detector | 248 |   1 |    1.0000 |    1.0000 |
       52 |                         Step counter | 249 |   1 |    1.0000 |    1.0000 |
       53 |                 Step counter wake up | 231 |   1 |    0.0005 |   25.0000 |
       55 |                   Significant motion | 250 |   1 |    1.0000 |    1.0000 |
       57 |                         Wake gesture | 232 |   1 |    1.0000 |    1.0000 |
       59 |                       Glance gesture | 234 |   1 |    1.0000 |    1.0000 |
       61 |                       Pickup gesture | 233 |   1 |    1.0000 |    1.0000 |
       63 |                 Activity recognition | 235 |   1 |    1.0000 |    1.0000 |
       67 |                   Wrist tilt gesture | 162 |   1 |    1.0000 |    1.0000 |
       69 |                   Device orientation | 163 |   1 |    1.0000 |    1.0000 |
       70 |           Device orientation wake up | 164 |   1 |    1.0000 |    1.0000 |
       75 |                    Stationary detect | 161 |   1 |    1.0000 |    1.0000 |
       77 |                        Motion detect | 160 |   1 |    1.0000 |    1.0000 |
       94 |                Step detector wake up | 230 |   1 |    1.0000 |    1.0000 |
Enable monitor..
[ 11189][I][SensorBHI260AP.hpp:650] configure(): Enable Accelerometer passthrough at 100.00Hz.
[ 11199][I][SensorBHI260AP.hpp:650] configure(): Enable Gyroscope passthrough at 100.00Hz.
Product ID     : 89
Kernel version : 5991
User version   : 5991
ROM version    : 5166
Power state    : active
Host interface : I2C
Feature status : 0x4a
Boot Status : 0x38: 	No flash installed. 
	Host interface ready. 
	Firmware verification done. 
Virtual sensor list.
Sensor ID |                          Sensor Name |  ID | Ver |  Min rate |  Max rate |
----------+--------------------------------------+-----+-----+-----------+-----------|
        1 |            Accelerometer passthrough | 205 |   1 |    1.5625 |  400.0000 |
        3 |           Accelerometer uncalibrated | 203 |   1 |    1.5625 |  400.0000 |
        4 |              Accelerometer corrected | 241 |   1 |    1.5625 |  400.0000 |
        5 |                 Accelerometer offset | 209 |   1 |    1.0000 |    1.0000 |
[... 6 lines cut ...]
       15 |                    Gyroscope wake up | 194 |   1 |    1.5625 |  400.0000 |
       16 |       Gyroscope uncalibrated wake up | 195 |   1 |    1.5625 |  400.0000 |
       28 |                       Gravity vector | 247 |   1 |    1.5625 |  400.0000 |
       29 |               Gravity vector wake up | 198 |   1 |    1.5625 |  400.0000 |
       31 |                  Linear acceleration | 246 |   1 |    1.5625 |  400.0000 |
       32 |          Linear acceleration wake up | 197 |   1 |    1.5625 |  400.0000 |
       37 |                 Game rotation vector | 252 |   1 |    1.5625 |  400.0000 |
       38 |         Game rotation vector wake up | 200 |   1 |    1.5625 |  400.0000 |
       48 |                        Tilt detector | 236 |   1 |    1.0000 |    1.0000 |
       50 |                        Step detector | 248 |   1 |    1.0000 |    1.0000 |
       52 |                         Step counter | 249 |   1 |    1.0000 |    1.0000 |
       53 |                 Step counter wake up | 231 |   1 |    0.0005 |   25.0000 |
[... cut ...]
       67 |                   Wrist tilt gesture | 162 |   1 |    1.0000 |    1.0000 |
       69 |                   Device orientation | 163 |   1 |    1.0000 |    1.0000 |
       70 |           Device orientation wake up | 164 |   1 |    1.0000 |    1.0000 |
       75 |                    Stationary detect | 161 |   1 |    1.0000 |    1.0000 |
       77 |                        Motion detect | 160 |   1 |    1.0000 |    1.0000 |
       94 |                Step detector wake up | 230 |   1 |    1.0000 |    1.0000 |

```