lewisxhe/SensorLib #20: BHI260AP Only one device working when enabling 2 at the same time

federica-ven, 2025-02-26:
Hi, I'm trying to use the SensorLib library with 2 instances of BHI260AP, one configured via SPI and one configured via I2C, the two devices initialise without issues and I can see al the relevant information being printed in the `begin()` phase but when I attach two interrupts to them (trying to just stream quaternion data from both) only one of the two interrupts is ever called, in my case the I2C device. 

If I run simply with either one or the other, then everything works just fine. Is there anything I should be doing to handle multiple instances of the same device?

this is a sample of my controller class 

```c++
class IMUController {
public:
    IMUController(int irqPin) : irqPin(irqPin), isReadyFlag(false) {}

    void begin(bool useSPI) {

        
        bhy.setPins(IMU_RESET);
        bhy.setFirmware(bhy2_firmware_image, sizeof(bhy2_firmware_image));
        bhy.setBootFromFlash(false);
        Serial.println("Initializing Sensors...");
        
        if (useSPI) {
            spiInterface = true;
              irqPin = INT_IMU2;
              // pinMode(INT_IMU2, INPUT);

            if (!bhy.begin(SPI, SPI_SS_IMU, SPI_MOSI_IMU, SPI_MISO_IMU, SPI_SCK_IMU)) {
                Serial.print("Failed to initialize sensor - error code:");
                Serial.println(bhy.getError());
                while (1) { delay(1000); }
            }
        } else {
            irqPin = INT_IMU1;
              // pinMode(INT_IMU1, INPUT);
            if (!bhy.begin(Wire, BHI260AP_SLAVE_ADDRESS_L, I2C_SDA_IMU, I2C_SCL_IMU)) {
                Serial.print("Failed to initialize sensor - error code:");
                Serial.println(bhy.getError());
                while (1) { delay(1000); }
            }
        }
        
        Serial.println("Initializing the sensor successfully!");
        
        BoschSensorInfo info = bhy.getSensorInfo();
        info.printInfo(Serial);
        
        // attachInterrupt(digitalPinToInterrupt(irqPin), dataReadyISR, RISING);
    }
    
    void initializeSensor(float sample_rate, uint32_t report_latency_ms = 0 ) {
        quaternion.enable(sample_rate, report_latency_ms);
    }

    void handleInterrupt() {
        isReadyFlag = true;
    }
    
    void update() {
        if (isReadyFlag) {
            isReadyFlag = false;
            bhy.update();
        }
        
        if (quaternion.hasUpdated()) {
            quaternion.toEuler();
            quaternionData.roll = quaternion.getRoll();
            quaternionData.pitch = quaternion.getPitch();
            quaternionData.yaw = quaternion.getHeading();
            if (spiInterface) {Serial.print("SPI_IMU: ");}
            else {Serial.print("I2C IMU: ");}
            Serial.print(quaternionData.roll);
            Serial.print(",");
            Serial.print(quaternionData.pitch);
            Serial.print(",");
            Serial.println(quaternionData.yaw);
        }
    }
    
    QuaternionData getQuaternionData() const {
        return quaternionData;
    }
    
private:
    // static void dataReadyISR() {
    //     instance->isReadyFlag = true;
    // }
    
    // static IMUController* instance;
    volatile bool isReadyFlag;
    bool spiInterface = false;
    SensorBHI260AP bhy;
    SensorQuaternion quaternion{bhy};
    QuaternionData quaternionData;
    int irqPin;

};
```

in my main sketch what I do is this: 

```c++

// in setup()
  imu1.begin(false);  // I2C
  imu2.begin(true);   // SPI
  imu1.initializeSensor(50, 0);
  imu2.initializeSensor(60, 0);

  // Attach interrupt service routines
  attachInterrupt(digitalPinToInterrupt(INT_IMU1), imu1ISR, RISING);  
  attachInterrupt(digitalPinToInterrupt(INT_IMU2), imu2ISR, RISING); 
```

and my ISRs are just setting a flag, which I then de-select from the main `loop()` when it's triggered.

lewisxhe, 2025-02-27:
Have you checked the interface information printed by info.printInfo(Serial); of the two BHIs to see if they are one I2C and one SPI? If so, it means that they are initialized normally, and you need to check whether the interrupt is actually triggered.

I have not tried to use two BHI260s, but the method I wrote supports multiple BHIs. I don't have multiple BHIs at the moment, so I can't make any tests. I will try some later.

federica-ven, 2025-02-27:
The information seems standard for both of them. If I run them separately they both work and the interrupt is triggered. When I try to run them concurrently then only one interrupt triggers.

lewisxhe, 2025-02-28:
Because I don't have a BHI260AP with SPI interface, I used two BHI260AP with the same address (0x28) connected to different Wire ports to test the software. The test results showed no problem, both sensors were normal.

closed 2025-03-10 by lewisxhe