lewisxhe/SensorLib #21: cant compile when update to 0.3.0 from 0.2.6

ElMyggo, 2025-04-05:
**Hi i cant compile my project with version 0.3.0 (0.2.6 working) i get this error message:**
Im using Arduino ide 2.3.5

`In file included from c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/SensorPlatform.hpp:35,
                 from c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/ExtensionIOXL9555.hpp:33,
                 from c:\Users\xxx\Documents\Arduino\libraries\LilyGo-T-RGB-main\src/LilyGo_RGBPanel.h:24,
                 from C:\Users\xxx\Documents\Arduino\Round_display_AMS_four_screen_v1\Round_display_AMS_four_screen_v1.ino:1:
c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/platform/arduino/SensorCommArduino_SPI.hpp:45:33: error: variable or field 'spi' declared void
   45 | #define  writeBytes(txBuf,size) spi.writeBytes(txBuf,size)
      |                                 ^~~
c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/platform/arduino/SensorCommArduino_SPI.hpp:45:33: error: expected ';' at end of member declaration
c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/platform/arduino/SensorCommArduino_SPI.hpp:45:36: error: expected unqualified-id before '.' token
   45 | #define  writeBytes(txBuf,size) spi.writeBytes(txBuf,size)
      |                                    ^
c:\Users\xxx\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/MsgPack/MsgPackSerializer.hpp: In member function 'size_t ArduinoJson::V731PB22::detail::MsgPackSerializer<TWriter>::visit(ArduinoJson::V731PB22::JsonString)':
c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/platform/arduino/SensorCommArduino_SPI.hpp:45:33: error: 'spi' was not declared in this scope; did you mean 'sei'?
   45 | #define  writeBytes(txBuf,size) spi.writeBytes(txBuf,size)
      |                                 ^~~
c:\Users\xxx\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/MsgPack/MsgPackSerializer.hpp: In member function 'size_t ArduinoJson::V731PB22::detail::MsgPackSerializer<TWriter>::visit(ArduinoJson::V731PB22::RawString)':
c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/platform/arduino/SensorCommArduino_SPI.hpp:45:33: error: 'spi' was not declared in this scope; did you mean 'sei'?
   45 | #define  writeBytes(txBuf,size) spi.writeBytes(txBuf,size)
      |                                 ^~~
c:\Users\xxx\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/MsgPack/MsgPackSerializer.hpp: In member function 'void ArduinoJson::V731PB22::detail::MsgPackSerializer<TWriter>::writeInteger(T)':
c:\Users\xxx\Documents\Arduino\libraries\SensorLib\src/platform/arduino/SensorCommArduino_SPI.hpp:45:33: error: 'spi' was not declared in this scope; did you mean 'sei'?
   45 | #define  writeBytes(txBuf,size) spi.writeBytes(txBuf,size)
      |                                 ^~~
exit status 1

Compilation error: exit status 1`

ElectricDream5, 2025-04-05:
I had pretty much the same error after updating to Arduino IDE 2.3.5 this week. It seems the errors are related to SensorLib library 0.3.0, so I deleted it and reinstalled the previous version (0.2.6) and I am able to upload sketches onto my LilyGoT-Display Amoled Plus. Hope that helps.

lewisxhe, 2025-04-07:
For other versions that depend on this library, do not upgrade to v0.3.0 yet, as the difference is huge. I will update the dependencies later.

closed 2025-05-15 by lewisxhe