lewisxhe/SensorLib #23: Can't compile on 0.3.1 (error: 'init' is not a member of 'SensorPCF85063')

MikeyMoMo, 2025-04-10:
Seems to be a problem with this library.  I will try an earlier version.

The error message is in the title.  Just compile something using the LilyGo_AMOLED library.  I am compiling for T4S3 using sample code AdjustBrightness from the AMOLED library examples.

Do I have the right place that this is coming from?

```
D:\Arduino\libraries\LilyGo-AMOLED-Series\src\LilyGo_AMOLED.cpp: In member function 'bool LilyGo_AMOLED::beginAMOLED_191_SPI(bool)':
D:\Arduino\libraries\LilyGo-AMOLED-Series\src\LilyGo_AMOLED.cpp:603:31: error: 'init' is not a member of 'SensorPCF85063'
  603 |     _hasRTC = SensorPCF85063::init(Wire, boards->pmu->sda, boards->pmu->scl);
      |                               ^~~~

```

Thanks for all your hard work on your libraries to help us.

Mike

lewisxhe, 2025-04-10:
This has been fixed, you should have installed it via the arduino library manager.
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series/commit/5fb5840943f8c1a3b0bf66f8f078f13e5e618058

MikeyMoMo, 2025-04-10:
Great.  Thanks.  And thanks for the very quick response!!

Mike

closed 2025-04-10 by MikeyMoMo