lewisxhe/SensorLib #25: GT911 Chip ID

jniebuhr, 2025-09-03:
I am using the GT911 touch chip on an ST7701S panel that's connected to an ESP32-S3. There is also an TCA9555 io extender on the same I2C bus. When the GT911 initializes it will not return 911 for some panels but instead just 9. On another board with a similar setup but a TCA9554 on the I2C bus the same panel works.

I have changed the device id to also allow "9" as the Chip ID and the chip is fully functional afterwards. Any idea on how to fix this?

jniebuhr, 2025-09-04:
Oh I just realized you're actually from LilyGo. So this happens with some of your 2.8" screens and did lead to similar problems that are described here https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/issues/70
On other screens from you it works just fine (Same panel, same PCB version, even same month of manufacturing)
The GT911 is not recognized and the driver defaults to the 2.1" initialization messages not only making the screen show gibberish but burning in some artifacts into the panel (they do fade out over time).

Steps I had tried:
- Manually activating the reset pin
- Lowering the I2C data rate to 100 kbps
- Adding in some delays

lewisxhe, 2025-09-04:
T-RGB please use the fixed version 0.2.3
https://github.com/Xinyuan-LilyGO/LilyGo-T-RGB/blob/359c03f7cd8e9aaadd4bf0f3fbf790844fd95f98/platformio.ini#L70